esp-rs/espup #517: espup install fails with 400 Bad Request when querying GitHub API for latest Xtensa Rust version

adsgodlove, 2025-08-01:
Hi there,

I'm encountering an issue when trying to install the Espressif Rust ecosystem using `espup`. The installation fails when querying the GitHub API for the latest Xtensa Rust version, returning a 400 Bad Request error. Here are the details:

### Steps to Reproduce
Run the command:
```bash
espup install -l debug
```

### Expected Behavior
`espup` should successfully fetch the latest Xtensa Rust version from the GitHub API and proceed with the installation.

### Actual Behavior
The installation fails with the following error:
```
[warn]: Failed to get latest Xtensa Rust version: HTTP GET Error: GitHub API returned status code: 400 Bad Request
Error: espup::toolchain::http_error

  × HTTP GET Error: GitHub API returned status code: 400 Bad Request
```

### Debug Logs
Relevant debug logs show repeated attempts to connect to the GitHub API:
```
[debug]: Querying latest Xtensa Rust version from GitHub API
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
```

### Additional Information
- When I manually query the same GitHub API endpoint using `curl`, it returns a valid response with the latest release information (v1.88.0.0).
- My environment: macOS (using a MacBook Pro)
- `espup` version: Not explicitly confirmed, but the latest available via crates.io (as the tool successfully fetches data from crates.io earlier in the process)

It seems like the issue might be related to how `espup` is forming or sending the request to the GitHub API, as the endpoint works correctly with a direct `curl` call. Could this be due to missing headers (e.g., User-Agent) or authentication requirements?

Let me know if you need any additional details to investigate this!

Thanks

JoelDavisEngineering, 2025-08-11:
> A few questions/suggestions:
> 
> * You could verify the `espup` version with `espup --version`.
> * Do you have any proxy environment variable set up (`HTTP_PROXY`, HTTPS_PROXY`, `ALL_PROXY`...)
> * Can you paste the full debug log?

Hello. I also receive the same error as the OP.

**Details**

OS: maOS 12.7.6
Machine: Macbook Pro, Intel
espup version: 0.15.1
rustc version: 1.88.0

No proxy environment variable is setup.

Here is the full debug report when running 'espup install -l debug': 

[debug]: connecting to crates.io:443 at [2600:9000:2365:c800:c:7ed3:240:93a1]:443
[debug]: No cached session for DnsName("crates.io")
[debug]: Not resuming any session
[debug]: Using ciphersuite TLS13_AES_128_GCM_SHA256
[debug]: Not resuming
[debug]: TLS1.3 encrypted extensions: [ServerNameAck]
[debug]: ALPN protocol is None
[debug]: created stream: Stream(RustlsStream)
[debug]: sending request GET https://crates.io/api/v1/crates/espup/versions
[debug]: writing prelude: GET /api/v1/crates/espup/versions HTTP/1.1
Host: crates.io
User-Agent: ureq/2.12.1
Accept: */*
accept-encoding: gzip
[debug]: Chunked body in response
[debug]: response 200 to GET https://crates.io/api/v1/crates/espup/versions
[debug]: dropping stream: Stream(RustlsStream)
[info]: Installing the Espressif Rust ecosystem
[debug]: Querying latest Xtensa Rust version from GitHub API
[debug]: Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[debug]: starting new connection: https://api.github.com/
[warn]: Failed to get latest Xtensa Rust version: HTTP GET Error: GitHub API returned status code: 400 Bad Request
Error: espup::toolchain::http_error

  × HTTP GET Error: GitHub API returned status code: 400 Bad Request

zpeters, 2025-09-22:
installed espup using `cargo install espup --locked`

rustc version = rustc 1.90.0 (1159e78c4 2025-09-14)
OS version / info = Darwin Zachs-iMac-6.local 24.6.0 Darwin Kernel Version 24.6.0: Mon Jul 14 11:28:17 PDT 2025; root:xnu-11417.140.69~1/RELEASE_X86_64 x86_64
espup version = espup 0.15.1

please let me know if there is other info i can contribute

(devbox) zachpeters@Zachs-iMac-6 ~/D/alarm_bridge (main)> espup install
[info]: Installing the Espressif Rust ecosystem
[warn]: Failed to get latest Xtensa Rust version: HTTP GET Error: GitHub API returned status code: 400 Bad Request
Error: espup::toolchain::http_error

  × HTTP GET Error: GitHub API returned status code: 400 Bad Request

(devbox) zachpeters@Zachs-iMac-6 ~/D/alarm_bridge (main) [1]> espup install --toolchain-version 1.88.0.0
[info]: Installing the Espressif Rust ecosystem
Error: espup::toolchain::http_error

  × HTTP GET Error: GitHub API returned status code: 400 Bad Request

(devbox) zachpeters@Zachs-iMac-6 ~/D/alarm_bridge (main) [1]> espup install --toolchain-version 1.88.0.0 -k
[info]: Installing the Espressif Rust ecosystem
[info]: Checking Rust installation
[info]: Installing RISC-V Rust targets ('riscv32imc-unknown-none-elf', 'riscv32imac-unknown-none-elf' and 'riscv32imafc-unknown-none-elf') for 'stable' toolchain
[info]: Installing Xtensa Rust 1.88.0.0 toolchain
[info]: Installing Xtensa LLVM
[info]: Installing GCC (xtensa-esp-elf)
[warn]: Installation for 'LLVM' failed, retrying. Error: HTTP GET Error: 400 Bad Request
[warn]: Installation for 'GCC (xtensa-esp-elf)' failed, retrying. Error: HTTP GET Error: 400 Bad Request
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: HTTP GET Error: 400 Bad Request
[warn]: Previous installation of LLVM exists in: '/Users/zachpeters/Projects/alarm_bridge/.devbox/virtenv/rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225'. Reusing this installation
[info]: Creating symlink between '/Users/zachpeters/Projects/alarm_bridge/.devbox/virtenv/rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225/esp-clang/lib' and '/Users/zachpeters/.espup/esp-clang'
[info]: Installing GCC (xtensa-esp-elf)
[warn]: Previous installation of GCC exists in: '/Users/zachpeters/Projects/alarm_bridge/.devbox/virtenv/rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906'. Reusing this installation
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.88.0.0 toolchain
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: HTTP GET Error: 400 Bad Request
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.88.0.0 toolchain
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: HTTP GET Error: 400 Bad Request
[warn]: Failed to detect version of Xtensa Rust, reinstalling it
[info]: Uninstalling Xtensa Rust toolchain
[info]: Installing Xtensa Rust 1.88.0.0 toolchain
[warn]: Installation for 'Xtensa Rust' failed, retrying. Error: HTTP GET Error: 400 Bad Request

thread 'tokio-runtime-worker' panicked at /private/tmp/nix-build-espup-0.15.1.drv-0/source/src/toolchain/mod.rs:344:32:
called `Result::unwrap()` on an `Err` value: SendError { .. }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Error: espup::toolchain::http_error

  × HTTP GET Error: 400 Bad Request

SergioGasquez, 2025-09-26:
> if i install espup in the "normal" way (ie from the package manager (nix i think)) the espup install fails

The package manager version may be outdated, we don't maintain any package manager `espup` package, I would recommend using `cargo install`, `cargo binstall` or downloading the artifacts from the GH releases.